oresat/oresat-solar-hardware #25: Brown out on initialization

davidalay, 2020-08-31:
Upon initialization of the MPPT, the F0 browns out if a load is connected.  Current theory is that the LT1618 is pushing too much current, as the system defaults to max current and starving the 3.3V line.  The lt1618 enable pin gets stuck at ~1.7 V during the system brownout (threshhold is at 1V).  Perhaps a RC timer on that pin so that it takes time for the LT1618 to enable, giving the the F0 time to boot.

andrewgreenberg, 2020-08-31:
I'm not sure what you mean "upon initialization of the MPPT"? 

Initializing the MPPT: When the M0 boots, the LT1618-ENABLE should be held low, disabling the LT1618, until the DAC is initialized and everything is ready. Then we SLOWLY bring up the LT1618 and go into MPPT mode. If the solar array voltage ever crashed below 4V, then we should stop MPPT mode and wait until we have more sunlight.

MPPT mode: If there's no load, the MPPT is going to immediately drive to 8.4V and sit there. If you attach a big load, suddenly, it might brown out because it doesn't have time to react. We always expect there to be batteries in the loop here, so the MPPTs don't have to deal with sudden loads.

davidalay, 2020-08-31:
It turned out that we were not holding the enable pin low for long enough to give the DAC time to initialize.  We implemented that and it seemed stable.  I am leaving the issue open for now though because I am not yet convinced we fully characterized/solved the problem.

What was happening is that when the array went from high irradiance, to low irradiance, back to high irraidance (simulating a BBQ roll) the system would brown out if any load was connected (this was with batteries on the bus), with the enable pin stuck at 1.7V.  If the load was removed, then replaced while the module was illuminated, the M0 would begin tracking normally.  We didn't think to measure cell voltage in the low irradiance mode, but AFAIK we don't have any software under voltage lockout on the lt1618 right now. 

It is possible that the cell was illuminated enough from the ambient room light (we didn't think to fully darken them, we just turned off our giant halogen) that the tps63070 was able to keep the M0 partially powered, hence the enable pin being stuck 'high' (the cutoff for the tps63070 is 2V, which I believe the alta cells sit higher than that at under ambient lighting conditions). 

I also think you are right about the MPPT not having time to keep up.  Our refresh rate was set arbitrarily slow so we could follow the tracking, so its possible that the M0 didn't have time to throttle the lt1618 before a brown out occurred.

davidalay, 2020-09-07:
This problem still persists when the light is dimmed and then brought back up.  The F0 does not reset, it gets 'stuck' with the 3.3V line not coming back up properly.  This is potentially a major issue

andrewgreenberg, 2025-01-31:
This is not an issue in the v5.3 or later, or v1.0 Si designs anymore. Not sure what this issue is, but we've never seen it. So I'm closing this issue.

closed 2025-01-31 by andrewgreenberg