oresat/oresat-solar-hardware #53: Bring up issues on the Silicon v1.0 design

andrewgreenberg, 2025-01-09:
Lots of small fixes:

- U4 (INA226) has the SDA and SCL lines swapped
- U4 (INA226) uses 100 ohm filter resistors, and should use 10 ohm or less. Also, see the datasheet, throw a capacitor on it.
- U2 and U11 (TMP101) are just... weird. Still! It's like a curse on this idea. Make the symbol match the pin out in the datasheet. Then fix the footprint to match, and make sure that the RIGHT PIN ONE is flagged. It's current wrong on the footprint. Also, while you're fixing the footprint, consider making that slot narrower to better grab the little SOT-23-6 feet.
- Q1 is in backward. Furthermore, there's is NO reason it needs to be a DMP3050. It's not like we use that anywhere else? So switch to a SOT-23 P channel transistor like our standard NTR4101.
- See the rest of the issues; I think they have very valid points, especially adding a separate shunt resistor for the INA226 and the LT1618. Sharing them is cute, but might be super noisy for the INA226.
- Add a white number area at LEAST on the component surface, but a small one on the side on the cell surface would be great.
-

andrewgreenberg, 2025-01-09:
Other notes: 

- You can officially kill TP27, TP26, and TP28, TP29

JackkP, 2025-01-10:
Additional Comment, need to change the PN for the smaller cells in the middle from KXOB2502X8F to KXOB2502X8**T**F the difference being cell thickness. New PN is the thin revision from anysolar.

JackkP, 2025-01-29:
- ✅U4 (INA226)
    - ✅SDA and SCL connected to correct pins.
    - ✅Connected Vbus to Vin- (DS pg. 28).
    - ✅Reduced R36 and R37 to either zero ohms or 10 ohms. Added a 100n capacitor across them (DS pg.14).
    - 🛈 **Did not change footprint, current footprint matches datasheet recommended landing pattern (DS pg. 38).**
- ✅U3 (LT1618)
    - ✅Removed resistors R34 and R35.
    - ✅Check footprint, switched to oresat-ics:MSOP-10_ms to match recommended landing pattern
    - ✅Killed TP5
- ✅U2 and U11 are fixed
    - 🛈 Did *not update oresat-ics:DBV6-SLOT footprint (to adjust width). instead, changed the pins in the schematic symbol oresat-ics:TMP101NA; unclear why this was wrong, probably a symbol library conversion error. The pins in the DBV6-SLOT are correct for a slotted inverted IC, but the pins in the schematic symbol were also flipped.
- ✅Switched Q1 to NTR4101P.
- ❓Did not add a separate shunt for the INA226... can probably make space to do that if we think it's worthwhile. Do we want to do this?
- ✅Added white serial number areas on component surface and on cell surface
- ✅Updated  KXOB2502X8F to KXOB2502X8**T**F

These changes are resolved in [5202ddf](https://github.com/oresat/oresat-solar-hardware/commit/5202ddf4df5e114483215eab15eb4e8ec2a75c87)

**Things still needing review:**

- ❓Why is the refdes for the M55 connector CF1?
- ❓Is the white serial number area on the cell side satisfactory?
- ❓Did we figure out why V1.0 boards 1 and 6 have 💩 power output?
- ❓Add separate shunts for INA226 and `LT1618?`
- ⚠I need to thoroughly review the board before we do a design review and look for bad stuff

andrewgreenberg, 2025-01-31:
- Yeah, I think you really can kill those TPs. What do you want to review about it?
- CF1 stands for "Connector, Female". Feel free to change to "J"
- White serial number area is too big. Make it more "square". Big enough to hand-write a two digit number.
- We don't yet know about #1 and #6, will have to investigate, but that shouldn't slow down the v1.1 design.
- No, let's skip the separate shunts. I like sharing them until we figure out that sucks.
- OK, do the last minute review, I think we're good to go to try v1.1!

andrewgreenberg, 2025-02-02:
Just did a design review on these changes - https://docs.google.com/document/d/1YUit9oMjWgY5fHvdJ-Ph0KPl91w79ojY2p2lc8lckxo/edit?usp=sharing

JackkP, 2025-03-04:
- ✅Added 4.7uF caps to LTC4411s on solar cells
- ✅Vbus now goes to Vin+ on the INA226 (to calculate power from cells before rather than after shunt)
- ✅Cleaned up silkscreen
- ✅Added 3 ground test points (fix for https://github.com/oresat/oresat-solar-hardware/issues/47)
- ✅Removed TP26, TP27, TP28, TP29
- TMP101 Changes
     - ✅Added TMP101 datasheet to symbol
     - ✅Fixed TMP101 footprint to be 2.3mm wide
          - Increased length of pad by 0.05mm (inwards) to account for going from 2.4 to 2.3mm slot width... can we check the pads to see if they look good in DR?
     - 🛈 Updated symbol for TMP101 to use KiCad built in symbol
     - 🛈 Updated TMP101 foootprint pinout such that symbol matches IC and slotted footprint matches upside-down IC
- ✅Updated TCAN330 footprint to Package_TO_SOT_SMD:SOT-23-8
     - Pad size is slightly different, 1.35x0.5 instead of 1.1x0.45, but soldermask openings are smaller and pad corners are rounded... I think it should be better
- ✅Cleaned up all traces
- ✅Added 3 ground test points (fix for https://github.com/oresat/oresat-solar-hardware/issues/47)
- ✅Added 3D models for all missing components

JackkP, 2025-03-04:
Remaining Todos:

Todos:
- ⚠Move VSOLAR testpoint to be accessible from top surface
- Export 3D model to Onshape?
- ...?

Review: **THIS IS NOT EXHAUSTIVE (there's no such thing); IF YOU ARE REVIEWING LOOK FOR OTHER GARBAGE**
- Cells page:
     - Check Solar Cell LTC4411 circuits
          - Are caps on inputs and outputs correct?
          - Pinout check
          - Are the caps in the schematic that are associated with each ideal diode the correct ones in layout (mega OCD)
          - Check bypass R, input, output cap routing for all ideal diodes
     - Check silicon cells part numbers
          - Make sure that part numbers are for thin cells
          - Double check DPN
               - If anything is wrong make sure to update for ALL numbered library items
- Power page:
     - Check INA226
          - SCL/SDA
          - Vin sense pins
          - 10 ohm resistors ok?
          - Current sense routing & capacitor placement look good?
     - LT1618:
          - Check pinout
          - Check connections to shunt resistor
          - Check enable and Iadj signals
     - Ideal Diode
          - Is the Q1 pinout correct
          - Are all Q1 PNs correct?
          - Double check U12 pinout
     - No changes to 3.3V SPS, CF1, or RBF, but should check anyways
- MCU page:
     - Check the TMP101 pinout 1000 times
          - Check SCL and SDA!
          - Check if new symbol looks correct
          - Are there any other boards using the TMP101... if so, we should check those because the footprint was updated *INCLUDING PIN NUMBERS
     - Check TCAN330 CAN XCVR pinout
- Misc:
     - Check to make sure silkscreen SN box looks fine
     - Check all silkscreen text to make sure it's clear which component it references
     - Go through DRC 1e+40 times

JackkP, 2025-05-18:
Updating Todos from 3-30 design review + fixes from last commit + checking everything

Critical fixes: None
Important Fixes:
 - Moved the VSOLAR TP to a place accessible from the cell side of the PCB
 - Moved a do-no-good ground test point next to the 3.3v test point available on the cell-side of the PCB
 - Swaped 3D model of CF1 to actual version which is longerr
Whatever your mom fixes:
 - Unchecked “Exclude from simulation” for all of the test points and for the hardware
 - Flipped the 3D model of the TMP101 in the slotted footprint for funzies
Other To Do:
 - Last minute ERC + DRC cleanup, commit, and have Andrew fab.
      - Swapped R17 symbol to Kicad symbol

Checklist:
- ✅ Cells page:
     - ✅ Checked Solar Cell LTC4411 circuits
          - ✅ Caps on inputs and outputs are correct (technically don't need the 1uF cap on the input)
          - ✅ Pinout check
          - ✅ Caps in the schematic that are associated with each ideal diode match between layout and schematic (mega OCD)
          - ✅ Check bypass R, input, output cap routing for all ideal diodes
     - ✅ Checked silicon cells part numbers
          - ✅ Part numbers are for thin cells
          - ✅ Double checked DPN
          - **⚠ Worth noting that if we switched to a 4 cell per component configuration instead of 8 cell per component (could put two in series to maintain the ~4.5v output) we could reduce the number of LTC4411's by a factor of 2 and slightly increase power by 10-20mW (since more of each SMD cell is actually covered with solar cell)**
- ✅ Power page:
     - ✅ Check INA226
          - ✅ SCL/SDA
          - ✅ Vin sense pins seem to be configured correctly
          - ✅ 10 ohm resistors good
          - ✅ Current sense routing & capacitor placement looks good
     - ✅ LT1618:
          - ✅ Pinout looks good
          - ✅ Connections to shunt resistor look great
          - ✅ Enable and Iadj signals look fine
          - ✅ FB resistor values look good
     - ✅ Ideal Diode
          - ✅ Q1 pinout is correct
          - ✅ Double checked U12 pinout
- ✅ MCU page:
     - ✅ Check the TMP101 pinout 1000 times
          - ✅ Check SCL and SDA!
          - ✅ Check if new symbol looks correct
     - ✅ Checked TCAN330 CAN XCVR pinout
- Misc:
     - ✅ Checked to make sure silkscreen SN box looks fine
     - ✅ Checked all silkscreen text to make sure it's clear which component is referenced
     - ✅ Went through DRC 1e+40 times
     - ✅ Updated REVISION and ISSUE_DATE variables

**All Changes are reflected in <insert commit>**

andrewgreenberg, 2026-01-07:
This all works now! Nice work Jack. Closing this issue.

closed 2026-01-07 by andrewgreenberg